blemasle/arduino-mcp23017 #4: INPUT-OUTPUT-Definition

Bms7, 2019-02-19:
In MCP23017.cpp line 41 setting of OUTPUT is:
iodir = readRegister(iodirreg);
if(mode == OUTPUT) iodir |= _BV(pin);
else iodir &= ~_BV(pin);

but from datasheet:
3.5.1 I/O DIRECTION REGISTER
Controls the direction of the data I/O.
When a **bit is set**, the corresponding pin becomes an **input.** 
When a **bit is clear**, the corresponding pin becomes an **output**.

So please correct
Thanks

blemasle, 2019-02-19:
Hi @Bms7 . You're absolutely right. Sorry about that. I just pushed the fix. The updated library should be available under the version `1.1.1` on the Arduino Library Manager within a few hours.

closed 2019-02-19 by blemasle

Bms7, 2019-02-19:
Hi,
in my version I simply changed  "OUT" of "OUTPUT" to "IN" -> "INPUT".
Was the easyiest way ;-)
Yours

 
 
 

repository","main_image_url":"https://github.githubassets.com/images/email/message_cards/header.png","avatar_image_url":"https://github.githubassets.com/images/email/message_cards/avatar.png","action":{"name":"Open 
in 
GitHub","url":"https://github.com/blemasle/arduino-mcp23017"}},"updates":{"snippets":[{"icon":"PERSON","message":"@blemasle 
in #4: Hi @Bms7 . You're absolutely right. Sorry about that. I just pushed 
hours."}],"action":{"name":"View 
Issue","url":"https://github.com/blemasle/arduino-mcp23017/issues/4#issuecomment-465176812"}}} 
"https://github.com/blemasle/arduino-mcp23017/issues/4#issuecomment-465176812", 
"url": 
"https://github.com/blemasle/arduino-mcp23017/issues/4#issuecomment-465176812", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"publisher": { "@type": "Organization", "name": "GitHub", "url": 
"https://github.com" } } ]

blemasle, 2019-02-19:
Yep, I struggled between the two solutions. Annnndd now I prefer the look of the other one (yours) 😄 Might change that on the next version...

pspastushkov, 2019-05-22:
Hi!
Problem with uint8_t MCP23017 :: digitalRead
"if (mcp23017_0x21.digitalRead(2)) " gives truth only when pin 0 is high
Changing "2" to another pin (1-15) does not give a result
How to read the state of pin number n?

Thanks

blemasle, 2019-05-22:
hi @pspastushkov. It should work. Could you try to directly read the register and ensure that the value is changing on your chip (use `readRegister(MCP2017_REGISTER::GPIOA)` or `GPIOB` ) ? If it is indeed not working, please open a new issue :)

pspastushkov, 2019-05-23:
Sorry.
Ok

commit 346e14d2 (2019-02-19): pinMode input/output logic was reversed. Fixed #4