blemasle/arduino-mcp23017 #5: delay time for a bounce switch

bill-in-iowa, 2019-04-30:
On the 23017 Arduino Library, I changed the delay statement for the switch bounce to checking the io pin according to the timimg on the data sheet instead of waiting the 200 milliseconds. For me the I2C is slow enough without waiting 200ms every time a switch is pressed

blemasle, 2019-05-01:
Hi,

Could you point me to what part of the code you're referring to ? The [only delay](https://github.com/blemasle/arduino-mcp23017/blob/7fe183a1a2f8e930fe424279f9639c04caf7ae78/examples/PortCopyOnInterrupt/PortCopyOnInterrupt.ino#L43) I can find in the library code is in the example.

It is of course an example that you can adapt as you wish depending on your needs :)

bill-in-iowa, 2019-05-01:
I'll look again, was thinking it was in the c++library . I've had to change several libraries that use delays as a way to time circuits. Thanks for replying so quickly

blemasle, 2019-05-13:
Any news on this @bill-in-iowa . Otherwise I'll close the issue :)

closed 2019-05-16 by blemasle