blemasle/arduino-mcp23017 #7: Possible to use TwoWire as software I2C

ScholliYT, 2019-07-02:
Can you give an example on how to use your lib with software i2c? In my project I got the i2c blocked for other parts and want to use your lib to controll a mcp23017.

https://github.com/blemasle/arduino-mcp23017/blob/346e14d2a55895da7a3a7337cceda7fe336ccc1c/src/MCP23017.cpp#L3

blemasle, 2019-07-04:
Hi @ScholliYT 
I never used SotwareI2C myself, but the tutorial at http://wiki.seeedstudio.com/Arduino_Software_I2C_user_guide/ should get you started.

Then simply pass your instance in the constructor. So my guess would be :
```cpp
#include "SoftwareI2C.h"

SoftwareI2C softwarei2c;
MCP23017 mcp = MCP23017(0x24, softwarei2c);

void setup()
{
    softwarei2c.begin(3, 2);       // sda, scl
}
```

ScholliYT, 2019-07-04:
Thank you 👍 I've got my problem fixed without software serial but I will try it out anyway.

blemasle, 2019-07-11:
Hi @ScholliYT . Have you had the opportunity to test the library with SoftwareI2C yet ?

ScholliYT, 2019-07-11:
Nope, sorry. Didn't have any time to test it. But it's still on my ToDo-List. Maybe you can give it a try yourself.

blemasle, 2019-07-11:
I moved home recently and I haven't had the opportunity to do some devs/electronics since then. No rush though :)

Thanks

SukkoPera, 2020-03-03:
Unfortunately this nice feature of this library cannot be used at the moment, due to a shortcoming of the original Wire.h library.

I have opened [this issue](https://github.com/Testato/SoftwareWire/issues/28) on a Software I2C library to detail the exact issue, so please see there.

There is already [an open issue](https://github.com/arduino/ArduinoCore-API/issues/31#issuecomment-594182230) on the Arduino Core. Everybody who is interested in this please add a thumbs up there.

blemasle, 2020-03-04:
Arf, that's a shame ! This as s kinda strange cause I didn't came with this idea out of the blue of course. I actually took it from another library, but I can't remember which one.

I'm not sure, but wouldn't a templated version of the library solve this issue ?

SukkoPera, 2020-03-04:
I've also realized it could be made to work using a template, but I see it more as a workaround rather than the proper solution.

So I would first try to get this solved in Wire.h by voting and commenting the issue I've linked above. Then, if we get nowhere with that, we can switch to a template. For my own needs, I have temporarily modified this library so that it uses that software implementation, and I can live with that for a while.

Beerlesklopfer, 2020-03-04:
Wooldn‘t it be nice being more general. The chip also has a SPI interface in the mcp23S17 version. When we think about software serial there could be also the hardware and software SPI option too; either in different constructors or a special class.

SukkoPera, 2020-03-04:
That could be interesting, but I think it should be filed as a different Feature Request. At the moment this is a library for the i2c version of the chip, you are asking to turn it into one for both the i2c and SPI versions. It's also quite a bit of work, actually :).

Beerlesklopfer, 2020-03-04:
Sure, but if we talk about concepts, we should have that in focus...

blemasle, 2020-04-16:
Until [ArduinoCore API #31](https://github.com/arduino/ArduinoCore-API/issues/31) is solved, nothing can be done here. A templated class isn't a solution either cause Arduino C++ does not support type constraints based on [concepts](https://en.cppreference.com/w/cpp/language/constraints).

closed 2020-04-16 by blemasle