blemasle/arduino-mcp23017 #11: Not able to initiate the class in an array

jviseur, 2019-09-27:
I have a number of MCP23017 and rather than initialising each individually I attempted to use the following:

      MCP23017 mcpi[4];
      int address = 0x20;
      for( int j = 0; j < 4; j++)
      {
        mcpi[j].MCP23017(address)
        address++;
      }

but get the error "no matching function for call to 'MCP23017::MCP23017()'

Are you able to confirm thus can be done and if so what am I doing wrong.

blemasle, 2019-09-27:
Hi.

`MCP23017 mcpi[4];` declares and tries to instantiate 4 `MCP23017` instances by calling the default constructor, which this class does not have. This is why you're getting the `no matching function for call to 'MCP23017::MCP23017()` error.

`mcpi[j].MCP23017(address)` makes no sense :) A constructor is meant to *creates* an instance, not to be called on it.

To achieve what your looking for, you can either define your array inline
```cpp
int address = 0x20;
MCP23017 mcpi[4] = {
  MCP23017(address++),
  MCP23017(address++),
  MCP23017(address++),
  MCP23017(address++)
};
```
Which I agree isn't really pretty
Or (although not recommended) use pointers

```cpp
MCP23017* mcp[4];
int address = 0x20;
for(int i = 0; i < 4; i++) {
  mcp[i] = new MCP23017(address++);
}
```

It would arguably be easier for you if the default constructor was present on the class. But on the other hand, I think in most cases it would confused users as the address would need to be set later on  and the library would fail at runtime.

jviseur, 2019-09-27:
Thank you for your answer.  “Or (although not recommended) use pointers” why do you not recommend using the pointer method?

The reason for asking is that the current 4 is only a proof of concept because in the final product will need to use two chains of 8 MCP23017 through a multiplexor.  The first 8 are inputs and the second 8 are outputs

blemasle, 2019-09-28:
Using pointers means you're allocating memory on the fly onto the heap rather than the stack. And as a general practice on such platforms, you should try to not rely on the heap as much as possible to avoid possible issues when you're heap collapses with the stack. See [this link](https://learn.adafruit.com/memories-of-an-arduino/optimizing-sram) for more info on that matter. 

On the other hand, your variables are static, allocated at start and will never be deallocated. So in the end, these allocations will never fragment the heap and leave some unused space. Maybe a pointer array is the way to go on this case after all :)

You could also use a second I2C bus to overcome the 8 chips on a bus limitation. This could either be an hardware bus or a software implemented bus (never used it myself, see [here](https://github.com/blemasle/arduino-mcp23017/issues/7#issuecomment-508369496)) tied on two digital pins depending on your platform capabilities.

jviseur, 2019-09-28:
Thank you @blemasle much appreciated.  I have attached a rough diagram of what we are planning to do.  It is for a model railway system and might up needing 100+ push buttons.   Currently all that is connected to the Arduino is Vcc, Grnd, SLC, SDA and an interrupt pin.   The whole process will be driven of the interrupt.   I am a software developer but have never as yet programmed in C++.   I have done a fair amount of coding in C# but have been mainly a Pick developer.

If you have any comments that will be greatly appreciated.

[image]

blemasle, 2019-09-29:
I can't really vet your design, I myself am not an electronic engineer at first :) I've never tried to multiplex i2c signal like that but I don't see a reason why it wouldn't work. You'd have to check every MCP23017 when an interrupt rise though, but that's not too much of an issue.

I think you have to try it on a breadboard and see for yourself if and how it works. That's what the fun is all about !

jviseur, 2019-09-29:
@blemasle that is the very reason I wanted to put the MCPs in an array as it is much cleaner code to iterate through them.  The other reason is that I can mostly use number in the array processing.

An other option is to read all and place in an 128bit unsigned number and then perform the compares.  I do not know if that is possible but I suspect / hope it is..

Thank you again for your assistance it is very much appreciated.

blemasle, 2019-09-30:
Well, you won't have a 128bit unsigned number, but you can easily put all your byte in a byte array and read it sequentially for sure :)

You're welcome. I'll close this issue for the time being. Do not hesitate to come back with questions regarding the library on another one. It could help others as well.

closed 2019-09-30 by blemasle

jviseur, 2019-09-30:
Thank you for the suggestion