blemasle/arduino-mcp23017 #13: Understanding Example PortCopyOnInterrupt

frankhoffmanndo, 2020-02-12:
It seems that I miss something in the PortCoypOnInterrupt Example.
My setup uses an Arduino MEGA in conjunction with the MCP23017

I attached buttons to port B (input port) that connect the MCP pins on port B to ground when pressed.
Port A pins serve as outputs for LEDs.

When I run the example and I press any of the buttons the corresponding pin on portA turns on.
The interrupt routine has been called and the interrupt has been handled.

However, after the first interrupt no further interrupts occur anymore, neither on the same pin nor on any other input pin.

As I understand the default input mode is with pull up, so after releasing the button (disconnecting ground) the input pin should return to high. 

Any idea why subsequent interrupts are ignored or don't get through?

frankhoffmanndo, 2020-02-12:
I am not sure if I connected the interrupt PIN correctly. I connected pin 3 of the Arduino (interrupt number 1) to INTB of MCP23017, INTA of MCP23017 is open.

blemasle, 2020-02-13:
Hi,

To me, I sounds like `clearInterrupts` isn't working properly or called. Without clearing the interrupt registers, any further change to the input pins won't trigger another interrupt.

Do you have a scope or another way to look at your interrupt pin (`digitalRead` should do it I think) to ensure that your pin 3 returned to `HIGH` state after your interrupt is triggered/processed ?

frankhoffmanndo, 2020-02-14:
Hi, 
I analyzed the problem, it is caused by the fact that I do not use a switch as input but a push button. If the time for pressing the bush button is larger than the delay time of 100ms, the interrupt is not properly cleared. Might have to do with the fact that the push button bounces. Your code was p
probably not meant to work with push buttons. I will fix it and make sure that the input port returns to high before clearing the previous interrupt.

blemasle, 2020-02-16:
I did use push buttons, but you might need to adapt the delay depending on how your buttons bounce. But 100ms is supposed to be a pretty generous delay, I'm surprised you'd need more than that.

You might also consider debouncing the buttons using dedicated hardware.

closed 2020-02-16 by blemasle