blemasle/arduino-mcp23017 #16: According to the issue SoftwareSerial: Make the bus more general => SPI / I2C (HW and SW)

Beerlesklopfer, 2020-03-04:
There is also a SPI version of that chip. Why not implementing that chip also?

blemasle, 2020-03-10:
Hi,

Sorry for the delay. While I'd like to see an SPI version of this library, I've never played with SPI myself and I don't have an MCP23S17 at hand to develop those modifications.

If you do, I strongly encourage you to give it a try by modifying the library code. On the top of my head, a base class with high level implementation and two classes for `MCP2017` and `MCP23S17` communication protocol with both signatures of `writeRegister` and `readRegister` might do the trick :)

waltje, 2025-09-04:
I just completely re-worked the library, added the 23S17 (and, thus, SPI mode), and will also add the 23008/9 8-bit models later.