blemasle/arduino-mcp23017 #17: _BV macro missed compiling for Arduino 33

andreacer, 2020-04-08:
Compiling on Arduino IDE 1.8.12 for Arduino 33, seems to be necessary to add

#define _BV(bit) (1 << (bit)) in the MCP23017.h .

andreacer, 2020-04-09:
Compiling on Arduino IDE 1.8.12 for STM32, it is a conflict between symbols GPIOA, GPIOB e some macros defined in the STM32DUINO shell. So if you need to make the lib portable, rename GPIOA -> GPIO_A and so on.

blemasle, 2020-04-09:
Hi @andreacer,

Thank you for the feedback. A better solution would rather be not to use CAPS in enum definitions to avoid conflicts with macros in general.

blemasle, 2020-04-16:
I'm having a hard time accepting to rename the registers to something that has meaning for the MCP23017. Currently the enum members match register names as defined in the datasheet.

`GPIOA` and `GPIOB` as macros is a stupid thing to do. Way too generic for such a common name :( They even declared `RTC`, `PWR` and `CRC` as macros. Couldn't be more generic/abusive

closed 2020-04-16 by blemasle

andreacer, 2020-04-17:
thanks for your work.