blemasle/arduino-mcp23017 #19: Multiple boards on 1 bus

PizzaProgram, 2020-12-15:
Is this library supports also multiple (max 8) boards too?  0..7
If yes, please write a few sentence to the Readme.md how to do that?

If not, please warn people of absence of this basic feature like this:

**NOTE:**
 - This library can handle **only 1 instance/board**, at default address x&000

... or something similar.

PizzaProgram, 2020-12-16:
- [ ] A picture from a board would increase the understandability for beginners what's this about
- [ ] And a [link to an "example of manufacturer" of a board version](https://www.waveshare.com/MCP23017-IO-Expansion-Board.htm)  also

# Why?
- Because it took me more hundreds of hours of searching and learning before I've found this library, _while I'm not a student_ .
- Because 1 picture can explain 1000 words, so it would help others a lot too.

[image]

blemasle, 2020-12-16:
Hi,

I'll add the available addresses as you suggested, thanks. Although this information is available in the MCP23017 datasheet (linked on the first line of the readme) that in my opinion you will need to read to use this library. The goal is not to have a library that you allow to play with an MCP23017 "at the surface", but rather an extensive control over the chip.

 > A picture from a board would increase the understandability for beginners what's this about
I have a different perspective on this :) A picture from a board would mislead potential users that this library is about using that board, not the chip itself.

> Because it took me more hundreds of hours of searching and learning before I've found this library
The library is the fourth result in the Arduino IDE for "mcp23017", first and second results for "mcp23017 arduino" on duckduckgo and google respectively.

So I cannot see what could be done to made it easier to find, or how what you've suggested would change something about it ¯\_(ツ)_/¯

> while I'm not a student
Not sure to get the point here, nor the tone or the implications of this sentence.

PizzaProgram, 2020-12-17:
< OFF >

**Hi,**

I'm happy you liked at least one of my suggestions.
( _I've thought I've explained myself good enough and you have realized how I'm trying to help for everyone first opening the main page of this project._ )

>Although this information is available in the MCP23017 datasheet (linked on the first line of the readme)

The goal would be to be as much informative at the first place as possible and to pass on ALL the important knowledge you guys know. 
I fully understand, that these things are "obvious" for you, but for us "mortals", it's mostly: "%+÷245˛˘˙1=()" ???
 :-D

That's where a few extra lines of code and a few pictures can help really really lot!

There are other half-solutions out there which can NOT handle multiple board on the same bus.
Nobody likes to deep-search for info, and nobody will download / fork until realized: 
 - YES! This is a great library! This can handle what I need!

That's all I'm trying to explain:
 - Just because YOU guys are professionals about the subject of this chip and this library, others are NOT!

How do I know?
 - Because I've made the same mistake! Skipped this project 3x.
 - Last time I was not sure >> opened a ticket to ask about, than 
 - edited later my question at first msg. to answer it myself, when I came back and saw the tiny hint that it CAN handle.

So a little bit of more documentation can not hurt. We are talking about +4 lines and a picture. 

> The goal is not to have a library that you allow to play with an MCP23017 "at the surface", but rather an extensive control over the chip

That's fine and I understand! 
( _Although beginners usually buy a board first, not just a chip. Experts know what to look for..._ )

>while I'm not a student 
Not sure to get the point here, nor the tone or the implications of this sentence.

I don't know what you mean. Sorry if my english is poor. I've wrote my first program in 1987, but still it's hard for me to understand all these information about IoT I've started to learn about 1 year ago. I can not even imagine how overvelming must be for example: for a student.
So I'm trying to suggest how to improve documentation, so others can learn faster.
And to advertise this great piece of work ! Because you guys are amazing. :1st_place_medal:

PizzaProgram, 2020-12-17:
Would you like this picture better?
[image]

blemasle, 2021-09-22:
I let this one out. Closing this, as I'm sorry to say I disagree with the initial statement. 

This library _will_ require you to read the chip datasheet. If you're not inclined to read a short readme intro and dive into the datasheet, but are rather looking for an image to find what you're looking at, then this library is not what you're looking. You're probably wasting your time and should look for a more high level api. 🙂

closed 2021-09-22 by blemasle