blemasle/arduino-mcp23017 #22: Problem with readPort

rhaamo, 2022-01-22:
I am using two mux (for the moment) to read a keyboard matrix, one mux uses all pins of both ports as outputs, the other one only 7 of port A as inputs (there is a diode from col to row on each key). The MCU is a teensy 4.1.

I am using a `Serial.println(mcpRows.readPort(MCP23017Port::A), BIN);` to check the activated rows but I am getting some weird output, at first it seems to shows the 8 pins, then when I start pressing a key it shows 6 digits, then maybe only one, 4, etc. any ideas what can be the issue ?

Code:
```cpp
#include <Arduino.h>
#include <Wire.h>
#include <MCP23017.h>

MCP23017 mcpCols = MCP23017(KBD_COL_MUX_ADDR, Wire1);
MCP23017 mcpRows = MCP23017(KBD_ROW_MUX_ADDR, Wire1);

void setup() {
  Serial.begin(115200);

  // Setup I2c (default ports)
  Wire.begin();
  // Setup I2c (second port)
  Wire1.begin();

  mcpCols.init();
  mcpCols.portMode(MCP23017Port::A, 0); // Port A output
  mcpCols.portMode(MCP23017Port::B, 0); // Port B output

  mcpCols.writeRegister(MCP23017Register::GPIO_A, 0x00);
  mcpCols.writeRegister(MCP23017Register::GPIO_B, 0x00);

  mcpRows.init();
  mcpRows.portMode(MCP23017Port::A, 0b11111111, 0b00000000, 0x00); // Port A input,no  pull up, ipolb

  mcpRows.writeRegister(MCP23017Register::GPIO_A, 0x00);
}

void loop() {
  /* Matrix buttons handling */
  for (int col = 0; col < 15; col++) {
    if (col < 8) {
      // Clear bank B
      mcpCols.writePort(MCP23017Port::B, 0x00000000);

      // Bank A
      mcpCols.writePort(MCP23017Port::A, muxPorts[col]);
      if (VERBOSE) {
        Serial.printf("COLA %d\t", col);
      }
    } else {
      // Clear bank A
      mcpCols.writePort(MCP23017Port::A, 0x00000000);

      // Bank B
      // Remove 8 to start from 0 on the second bank
      mcpCols.writePort(MCP23017Port::B, muxPorts[col-8]);
      if (VERBOSE) {
        Serial.printf("COLB %d\t", col);
      }
    }

    // Get rows status
    // byte rowStatus = mcpRows.readPort(MCP23017Port::A);
    if (VERBOSE) {
      Serial.println(mcpRows.readPort(MCP23017Port::A), BIN);
    }

    // For debugging purposes
    delay(10);
  }

}
```

example output:
```
COLA 0  10011000
COLA 1  10011000
COLA 2  10011000
COLA 3  10011000
COLA 4  10011100
COLA 5  10011100
COLA 6  11100
COLA 7  11000
COLB 8  11000
COLB 9  11000
COLB 10 11000
...
COLB 13 11001
COLB 14 11001
COLA 0  11001
COLA 1  11001
COLA 2  11001
COLA 3  11001
...
COLB 14 1000
COLA 0  1000
COLA 1  1000
COLA 2  1000
COLA 3  1001000
COLA 4  1001000
COLA 5  1001000
COLA 6  0
COLA 7  0
COLB 8  0
COLB 9  0
COLB 10 0
COLB 11 0
...
COLA 5  0
COLA 6  0
COLA 7  0
COLB 8  0
COLB 9  0
COLB 10 0
COLB 11 0
COLB 12 0
COLB 13 0
COLB 14 0
COLA 0  0
COLA 1  0
COLA 2  1000000
COLA 3  1000000
COLA 4  1000000
COLA 5  0
COLA 6  0
COLA 7  0
COLB 8  0
COLB 9  0
```

blemasle, 2022-01-23:
Hi,

I'm sorry, I have no idea on what's going on here. Did you press any key during that log output, or was it just a log in a loop without any interaction from your side ?

Also, you might want to use interrupts instead of continuously looping on reading values 🙂

rhaamo, 2022-01-23:
Hi, yes the output changed when I was pressing buttons.

I think I can try to add something like in `PortCopyOnInterrupt` that wouldn't be too complex while keeping column state info.

blemasle, 2022-05-09:
Closing this, feel free to reopen if needed 🙂

closed 2022-05-09 by blemasle