blemasle/arduino-mcp23017 #24: Encoders

bespezzano, 2022-05-09:
Hi, thanks very much for the library. Does it support rotary encoders? I can't figure out how to get a proper reading.

blemasle, 2022-05-09:
Hi,

Well with proper code, it supports anything MCP23017 are equipped to deal with 🙂.

Rotary encoders should pose no problem at all, but I never worked with them myself so I'm not sure how I can be of assistance. Reading rotary encoders with this lib should be no different from how you would do it straight from the arduino pins or from another chip.

bespezzano, 2022-05-09:
Thanks for the quick reply. Encoders can be a bit tricky, I think I have a timing issue. Have to track values, and that's where I think I'm getting muddled up. I understand the read() method gives me what's currently being pressed (which is probably too late for me), however I thought I could read the int registers to see what triggered the interrupt. Could you please clarify for me what's the difference between what you get with these 2 instructions?:
    mcp1.interruptedBy(a, b);
    mcp1.clearInterrupts(captureA, captureB) 

Thanks for your help.

blemasle, 2022-05-11:
Sure.
Have a look at [this example](https://github.com/blemasle/arduino-mcp23017/blob/master/examples/PortCopyOnInterrupt/PortCopyOnInterrupt.ino#L62)

`interruptedBy` gives you the pins that triggered the interrupt, not the value. Depending on your configuration, the interrupt might be triggered on active low, high or on change for instance.

`clearInterrupts` clear the MCP23017 interrupt flag so a new interrupt can be triggered. `captureA` and `captureB` contains the value of every pin the moment the interrupt was triggered.

Using both method this way minimize the risk of missing interrupts, while getting every bit of information out of the MCP23017 (what pin(s) triggered the interrupt and what were their values at the moment the interrupted was triggered).