blemasle/arduino-mcp23017 #31: Init method should not enable pull-up by default

rolgordijn, 2023-02-12:
The init() method enables the pull-up resistors by default. 
this means that there is 5V on each pin.  Even if I then configure the pins as output, there will be a very short time that a weak 5V is present at the output. 

This is really undesirable.  

In my case, my Arduino board is powered with 16V AC.  
This is then converted to DC, and from that DC voltage, I make 9 volts with a step-down converter and 5 volts with a linear regulator to power my circuit.  I'm controlling solenoids from that stabilized DC voltage. These are activated with a transistor connected to the MCP23017 with an appropriate base resistor. 

At power-on, the pull-ups are activated, activating all 16 solenoids. They draw to much current, causing the DC voltage to drop and this causes a brown-out reset. (and the cycle starts again. Arduino starts, executes init() method, activates all solenoids, brown out reset, ....)

blemasle, 2023-02-15:
Hi,

You're probably right that the pull-up should not be enabled by default, although I'm not sure if you shouldn't have some way of mitigating this issue in your circuitry anyway 🤔 .

Could you prepare a PR for this change ? I don't really have the time to do so at the moment 🙂

rolgordijn, 2023-02-15:
There are indeed ways to improve the power supply circuitry, but even then there are probably situations where connected electronics might come into an unknown state due to the pulses on the pins.

closed 2023-02-19 by blemasle