blemasle/arduino-mcp23017 #34: Interrupt behaviour on multiple MCP23017

bespezzano, 2023-11-17:
Hi,

I'm stuck with an interrupt issue and hoping to get some guidance.

I'm using 2 MCP23017 with an Arduino Mega on different interrupts and a couple of push buttons. MCPs are initialised in the same way, and the code to read them is the same, however the behaviour is slightly different. 

Here's the outuput after pressing and releasing 2 push buttons, one connected to each MCP:
```
20:00:10.626 -> MCP1 Interrupted
20:00:10.626 -> Port A -> 11111101
20:00:10.626 -> Port B -> 11111111

20:00:13.056 -> MCP4 Interrupted
20:00:13.056 -> Port A -> 11111101
20:00:13.056 -> Port B -> 11111111
20:00:13.148 -> MCP4 Interrupted
20:00:13.148 -> Port A -> 11111111
20:00:13.148 -> Port B -> 11111111
``` 

So looks like MCP1 is only catching when I push down the button, but not the release. Any idea what could be causing this difference in behaviour?

Here's my initialisation code:

```
    mcp1.init();
    mcp1.portMode(MCP23017Port::A, 0xFF);          
    mcp1.portMode(MCP23017Port::B, 0xFF); //Port B as input

    mcp1.interruptMode(MCP23017InterruptMode::Or);      
    mcp1.interrupt(MCP23017Port::A, FALLING);
    mcp1.interrupt(MCP23017Port::B, FALLING);
    

    mcp1.writeRegister(MCP23017Register::IPOL_A, 0x00);
    mcp1.writeRegister(MCP23017Register::IPOL_B, 0x00);

    mcp1.writeRegister(MCP23017Register::GPIO_A, 0x00);
    mcp1.writeRegister(MCP23017Register::GPIO_B, 0x00);

    mcp1.clearInterrupts();

     pinMode(MCP1_INT_PIN, INPUT_PULLUP);

    attachInterrupt(digitalPinToInterrupt(MCP1_INT_PIN), MCP1_int, FALLING);
```

And here's how I read the values:

```
    if(MCP1_interrupted) 
    {
      Serial.println("MCP1 Interrupted");

     MCP1_interrupted = false;
      mcp1.interruptedBy(a, b);
      mcp1.clearInterrupts(captureA, captureB);     

      currentA = mcp1.readPort(MCP23017Port::A);
      currentB = mcp1.readPort(MCP23017Port::B);

      Serial.print("Port A -> ");
      Serial.println(currentA,BIN);
      Serial.print("Port B -> ");
      Serial.println(currentB,BIN);
      mcp1.clearInterrupts();
    }
```

Cheers,

Bruno

bespezzano, 2023-11-17:
I've tried different MCPs and it seems the first one always has that difference in behaviour. E.g. replaced MCP1 (0x20) with MCP2 (0x21) with same results.

bespezzano, 2023-11-18:
Thanks very much for taking the time to reply :)

You are correct, there's no debouncing logic in that code as I implemented it on hardware for my project. That being said, these tests were not performed on that hardware, which would explain the second reading on MCP2/4, however that doesn't explain why it doesn't happen on MCP1 (or any "first" MCP connected, e.g. with the lowest address).

I will do some more testing on the actual hardware. I just found the behaviour discrepancy a bit odd and thought I'd mention it.

Cheers

blemasle, 2023-12-09:
Without further update, I'm closing this issue. Feel free to reopen it if you need more help 🙂

closed 2023-12-09 by blemasle